JensenNick/huawei_solar_pees #13: NRI Battery is completely wild

AndreasBroager, 2024-01-27:
I have just updated the file. 

I have set the state of both variables to zero in the dev UI and made a soft reset of HA. 

[image]

Is it already a fixed issue? and do I just need to reset some history?

JensenNick, 2024-01-27:
Hi Andreas! I'm looking into this at the moment. I'm afraid that the float(0) added the 22nd of January e5b5d11435fb0f420506532979287dfa3a9e1c0b has caused issues in the utility meters. Float(0) was added to please single inverter users. My double inverter install was stable all december.

JensenNick, 2024-01-27:
@AndreasBroager I belive the isse you have experienced now is fixed with this correction

JensenNick, 2024-01-27:
@AndreasBroager I belive the isse you have experienced now is fixed with this correction https://github.com/JensenNick/huawei_solar_pees/discussions/14 . Ill mark the post as fixed. Please let me know if the issue persist in your installation.

AndreasBroager, 2024-01-28:
I have now pulled thed latest one. Did a full restart of HA. 

[image]

Do i need to reset some history? If it is only history with the variables you have made, I don't care if I wipe them.

JensenNick, 2024-01-29:
@AndreasBroager Difficult for me to say because I don't know your "level" and I'm no expert either. The NRI Battery is based on this sensor. You have to identify which of the inputs for `sensor.pv_economy_nri_battery`  is behaving wrong / has caused the abnorm increase. It can't be the utility meters unless you have changed the input on the card you show. If you have done so please let me know, because things would be somewhat easier - but if not I'd like to get to the bottom of this.

 Please share the code you are using in the card and the full history graphs for the four input to this sensor.

```
- name: "PV Economy - NRI Battery"
        unique_id: pv_economy_nri_battery
        unit_of_measurement: DKK
        device_class: monetary
        state_class: total
        state: >
          {{ (
          states('sensor.battery_discharge_grid_sale_2') | float +
          states('sensor.battery_discharge_house_saving_2') | float -
          states('sensor.battery_charge_yield_sale_2') | float -
          states('sensor.battery_charge_grid_cost_2') | float
          ) | round(5) }}
        availability: >
          {{ has_value('sensor.battery_discharge_grid_sale_2') or
          has_value('sensor.battery_discharge_house_saving_2') or
          has_value('sensor.battery_charge_yield_sale_2') or
          has_value('sensor.battery_charge_grid_cost_2') }}
```
Looking at this I don't think this "bug" is solved unless you have changed something. I'll change the satatus.

JensenNick, 2024-01-29:
@AndreasBroager Please have a look at this https://github.com/JensenNick/huawei_solar_pees/issues/20 You're issue could very well be related to this. I'm running a test right now. The package file has been revised, but I'd like to have a little more "solid ground" before i announce it.

AndreasBroager, 2024-01-29:
I have updated to master/main again. it is still fubar. 
Mine looks like this below: 
```
      - name: "PV Economy - NRI Battery"
        unique_id: pv_economy_nri_battery
        unit_of_measurement: DKK
        device_class: monetary
        state_class: total
        state: >
          {{ (
          states('sensor.battery_discharge_grid_sale_2') | float(0) +
          states('sensor.battery_discharge_house_saving_2') | float(0) -
          states('sensor.battery_charge_yield_sale_2') | float(0) -
          states('sensor.battery_charge_grid_cost_2') | float(0)
          ) | round(5) }}
```

I don't have the `availability` part. But I don't know if it is something you have removed later on. 

I think I found the culprit.  But I am not sure how to fix it. 
[image]

JensenNick, 2024-01-29:
> I don't have the `availability` part. But I don't know if it is something you have removed later on.

If you go to the issue I've linked to, there is a link to the History log. Here you can see the changes made. I did remove the 'availability' and added 'periodically_resetting: false' to the connected utility meters.

> I think I found the culprit.  But I am not sure how to fix it. 

I hope you understand, that these are just sensors it's not an intelligent integration, so in order not to have issues all sensors should be deleted and purged before pasting the new package file into your HA.

JensenNick, 2024-01-30:
@AndreasBroager I have revised the PV sensor. They need availability and float without "(0)". Se correction here a5e58a13df62f4f08f80be8128e07870f125a987. Everything is running as it should except the issue with the usage of the `sensor. energy_import` and presumably `sensor.energy_export` in the Energy Dashboard. See here https://github.com/JensenNick/huawei_solar_pees/issues/22. The work as they should in the PEES package. I've posted on the HA forum and hopefully someone can help with a sollution to the Energy Dashboard.

[image]

[image]

closed 2024-01-30 by JensenNick