JensenNick/huawei_solar_pees #17: Error corrected - State Class for PV Economy Sensors

JensenNick, 2024-01-29:
### Discussed in https://github.com/JensenNick/huawei_solar_pees/discussions/15

<div type='discussions-op-text'>

<sup>Originally posted by **JensenNick** January 28, 2024</sup>
Error corrected: State class total_increasing corrected to total for all pv_economy sensors. Total_increasing is not supported for state_class: monetary.
History log: cc830b31fa3a558599f139141ed3010df1119e0e

[image]
</div>

JensenNick, 2024-01-29:
I'm onto something here. It seams like when you do an one time install HA is handeling out names in an way which I did not intend / anticipate. It seams like the error is provoked by that some sensor use the same prefix "_2" as HA does.

JacobsenKim, 2024-01-29:
min ser sådan her ud

[image]

Virker til den er gået død for mig den 26 omkring kl 6:00

JensenNick, 2024-01-29:
> Virker til den er gået død for mig den 26 omkring kl 6:00

Tror ikke ligefrem at det er samme issue, men jeg har også svært ved at se at PEES skulle være årsagen. Er du evt- løbet tør for diskspace i din database?

JensenNick, 2024-01-29:
> jeg har lige opdateret min fil med en frisk kopi fra siden, så må vi se hvordan den opføre sig. Jeg har kaldt min for # v0.0.6 29-01-2024 17:19

@JacobsenKim Jeg har opdateret package filen. Tror jeg har fået det fikset nu. Test kører på en helt fresh install af HA og ser rigtig ud.  Men det jeg har findet forklarer ikke rigtigt dine grafer - eller jeg kan i hvert fald ikke gennemskue hvad der er sket hos dig  kl. 06 den 26.

Jeg skal nok begynde at give versions numre, men vi er nødt til at komme over disse start vanskeligheder, der desværre overrasker mig lidt. Jeg har trods alt haft det kørende en måned på min maskine. Så kom der godt nok en rettelse med afladning af batteri til net og så burde jeg nok have haft en testmaskine sat op noget tidligere.

JacobsenKim, 2024-01-29:
Jeg aner ikke hvad der er sket men kan se den omkring kl 10 (hvor jeg sov) er begyndt at bevæge sig lidt men klart se en forskel efter jeg hentede en frisk kopi og indsatte el pris sensor samt invertere correction.

Jeg kan have været inde på filen og måske trykket på forkert keyboard og skrevet noget jeg ikke har set også senere gemt og igen senere genstartet ha, det er også lige meget, det ser ud til at være liv nu.

[image]

Ja kan godt se du har lavet om på sensor navne, det giver også mere mening at kalde sensor 2 for 2 end (), jeg bøvlede noget rundt i det da jeg kiggede på den sensor.

ang. test maskine så kan invertere ikke lide at tale med 2 ha maskiner på en gang. her har jeg haft en VM kørende fra ca. 17 til ca. 13

[image]

Jeg må igang igen med at opdatere :-D
Tænker at ligge 

> Efficiency Corrected Power Input Sensor #8

ind i en separat fil så jeg ikke skal bøvle med at kopiere den over hver gang, men ved ikke om det vil virke på den måde.

Super med v-nr. :-D

JensenNick, 2024-01-30:
@JacobsenKim Jeg har med de seneste revisioner rettet på ovenstående og lukker derfor issue. Tak for indspark.

[image]

closed 2024-01-30 by JensenNick