JensenNick/huawei_solar_pees #22: Energy Dashboard - Energy Import

JensenNick, 2024-01-30:
Using `sensor.energy_import` as input sensor in the Energy Dashboard has caused failure of the sensor 'Energy Import Cost' which HA has created. The input sensor is created as a platform sensor,.

[image]

JensenNick, 2024-01-30:
I've asked about the behavior of the sensor created by HA in the HA forum.
https://community.home-assistant.io/t/platform-integration-sensor-crashing-in-energy-dashboard/681769

JensenNick, 2024-01-30:
I have added Energy Utility Meters to all of the platform energy sensors. This seams to be the solution recommended by the HA community.

History log add7272704b160a4600d6a295004167158231a88

https://community.home-assistant.io/t/how-i-solved-a-reset-of-an-integral/540322/2

Roving-Ronin, 2024-01-30:
@JensenNick of your using the utility_meter and also wanting (it appears) to have the cost of all those sensors (import / export / battery and even different tariff periods), you'd be better looking at the energy_utility integration.

Works like the utility_meter but with a few more functions to allow defining from_grid or to_grid and calling a sensor with the energy cost (import or export) 

Easy as to implement just like a utility_meter but automatically creates all the cost / compensation sensors.

Add in tariffs and you end up with tracking for every single tariff type and cost (& and then add hourly/daily/weekly/monthly/quarterly/yearly versions). 
In the Energy Dashboard (or other cards) then allows easily seeing peak/shoulder/offpeak energy and costs and same for battery.

Also works for gas and water btw, not just electricity.

Roving-Ronin, 2024-01-30:
If of interest see: https://github.com/zeronounours/HA-custom-component-energy-meter

JensenNick, 2024-01-30:
> of your using the utility_meter and also wanting (it appears) to have the cost of all those sensors
No, the input for the Energy Dashboard is a platform: integral sensor. Nothing fancy at all and the input is from the Huawei Solar integration. I've tried to fix this without any success.
```yaml
  - platform: integration
    source: sensor.power_import
    name: "Energy Import"
    unique_id: energy_import
    round: 6
    unit_prefix: k
    unit_time: h
    method: left
```

JensenNick, 2024-01-31:
I will leave this "issue" as "open" but "locked", so it appears in the

JensenNick, 2024-10-05:
I haven't encountered this issue for the last 6 months. I'll close the issue permanently.

closed 2024-10-05 by JensenNick