JensenNick/huawei_solar_pees #58: Feature request: Enhanced Energy Management Dashboards

heinemannj, 2024-06-29:
You are looking at this time to raw Huawei Solar and PEES values which is helpful for debugging purpose.
Monitoring of all EMS dependant components in one view is basic and mandatory in future:

[image]

[image]

Merging is possible but only after heavy adaptation of sensor naming conventions ...
I strongly disagree on your actual sensor namings.

heinemannj, 2024-06-29:
I've made a quick and dirty review of your code.

My view:

- Very complex
- Hardly to maintain
- Not closely enough to @wlcrs naming convention

My conclusion:

- I will step out

**Sorry for the noise ...**

JensenNick, 2024-07-04:
No worries. I’m afraid we all have our own ide about naming. I’ve had no intention but to keep a unique naming to avoid any conflicts.

closed 2024-07-04 by JensenNick