JensenNick/huawei_solar_pees #59: Feature request: Enhanced Top Level Power/Energy Flow Dashboard

heinemannj, 2024-06-29:
Monitoring of all EMS Power/Energy Flows in one view is basic and mandatory in future:

[image]

Merging is possible but only after heavy adaptation of sensor naming conventions ...
I strongly disagree on your actual sensor namings.

heinemannj, 2024-06-29:
I've made a quick and dirty review of your code.

My view:

- Very complex
- Hardly to maintain
- Not closely enough to @wlcrs naming convention

My conclusion:

- I will step out

**Sorry for the noise ...**

JensenNick, 2024-07-04:
Presentation in Lovelace is very individual, but it would be great to biuld a base for people to start with. It has not been my main priority. It’s on the roadmap.

closed 2024-07-04 by JensenNick