JensenNick/huawei_solar_pees #60: Feature request: Forecast and Prediction EMS Dashboards

heinemannj, 2024-06-29:
You are looking at this time to raw Huawei Solar and PEES values which is helpful for debugging purpose.
In addition you are on the way to integrate historical data - Very good.

But only looking into the past and to the future on the same time can safe our Earth and our money ...

[image]

From my perspective the most complex task artifizielle Intelligenz (AI) / Machine learning (ML) - A lot of different integrations to be synchronized ...

heinemannj, 2024-06-29:
I've made a quick and dirty review of your code.

My view:

- Very complex
- Hardly to maintain
- Not closely enough to @wlcrs naming convention

My conclusion:

- I will step out

**Sorry for the noise ...**

JensenNick, 2024-07-04:
Agree, battery charge and discharge automations based on solar forcast and energy prices are key. Don’t forget your carharger. It is way beyond the scope of this project and my skills to create a universal automation which everyone can adapt. I’m working on my own setups in NodeRed, but they are based on my specific situation and setup.

closed 2024-07-04 by JensenNick