JensenNick/huawei_solar_pees #61: Feature Request: Enhanced Electricity Spot Price Monitoring

heinemannj, 2024-06-29:
I'm with Tibber and living in Germany:

[image]

[image]

[image]

[image]

[image]

heinemannj, 2024-06-29:
I've made a quick and dirty review of your code.

My view:

- Very complex
- Hardly to maintain
- Not closely enough to @wlcrs naming convention

My conclusion:

- I will step out

**Sorry for the noise ...**

JensenNick, 2024-07-04:
What is the issue implementing the pricing from Tibber?