JensenNick/huawei_solar_pees #65: Economy senors won't update anymore

pschneider87, 2024-08-22:
Hi,

first, thanks for the solution you provide here - it's streamlined to install and use and makes live so much easier!
However, I'm facing one issue I don't come to a solution with.

First:
- the Solar PV Dashboard is set up and working fine
- the Solar PV Grace View is set up and all sensors show data

[image]
[image]

I had to extend the Solar PV Dashboard to add my modell of Inverter with 20 kW
For this, I changed the input.yaml

`  inverter_1_3phase_models:
    name: "Inverter #1 3phase Models"
    options:
      - Select
      - SUN2000 3 KTL-M1
      - SUN2000 4 KTL-M1
      - SUN2000 5 KTL-M1
      - SUN2000 6 KTL-M1
      - SUN2000 8 KTL-M1
      - SUN2000 10 KTL-M1
      - SUN2000 20 KTL-M1`

and
`{% elif is_state('input_select.inverter_1_3phase_models','SUN2000 8 KTL-M1') %}
              {{ 8000 }}
            {% elif is_state('input_select.inverter_1_3phase_models','SUN2000 10 KTL-M1') %}
              {{ 10000 }}
            {% elif is_state('input_select.inverter_1_3phase_models','SUN2000 20 KTL-M1') %}
              {{ 20000 }}
            {% endif %}`

I set prices in the input dashboard to
30.27 (Import)
8.4 (Export)

Issue:
the economy sensors started to work at the first day, but are now stuck to their values. And while Values like for the Yield, Consumption and Battery change, all others remain as is.

[image]

Can someone help to fix this?

Roving-Ronin, 2024-08-22:
Hi @pschneider87 Phillip,

What version of the WLCRS / Huawei Integration are you running ?

I'm currently trying to work on updating PEES, to support the next release of the Huawei integration. It's the 'forthcoming release' as apart from what looks like adding Support for Huawei EMMA it also contains some extra sensors that I have got added into it, that allow pulling things like the inverter / batteries rated capacity / PMAX / Export Limitation of the inverters. Atop that there's some derived sensors that will work out the model of the inverter(s) .. combine that with rated power / pmax and you've got all you need to automatically configure PEES for the inverters, also work out how many batteries you've got installed.

Just need to work out the sensors and script / automations to auto detect the data needed to get rid of the manual selecting 1p or 3p inverters, what model and how many and how much battery is attached. All you then need to do are define the voltages you run them at / how many and of what size panels your running etc.

pschneider87, 2024-08-22:
Im running on 1.4.1

The upcoming changes sound great, though I'm not sure if this is the root cause of the issue here?

Maybe good to say that I had added the pees with DKK in the beginning, and switched them to EUR later. But this was shown in the statistics with a conversion as a fix

Roving-Ronin, 2024-08-22:
@pschneider87 the naming format for the inverters and battery all changed with v1.4.x

Previously iirc it was inverter_1_xxx and inverter_2_xxx
If you have a battery it was battery_name_of_sensor

With v1.4.x that all changed and it's now sensor.inverter_xx for #1 and # has _2 as a suffix at the very end of the entity.
Battery is now sensor.battery_1_xxx and (you have to enable it) there is also the new sensor.batteries_ that presents a bunch of new sensors that give the 'overall' status of all/multiple batteries.
Even if you have 1 battery the batteries sensors are better to use, and also the o es that I've added the se does that show the battery capacity (ie 5/10/15/20/25/30kW or the new 7/14/21... up to 84kWh.. though the WLCRS integration only supports up to 2 batteries per inverter atm fyi).

pschneider87, 2024-08-22:
@Roving-Ronin thanks. but I don't get it :-) can you point me where I need to change something? The thing is, all sensors are working fine, except the economical ones.

So I see data for the batteries, and the inverter in the Solar PV Grace View, but not for the economical sensors, which stay the same since days...

Roving-Ronin, 2024-08-22:
Sounds like your sensors haven't updated when going to v1.4.x so are still working with PEES atm. If you reinstalled WLCRS then you'd end up with the sensors I refer to (and that you'll want to move to in future).

pschneider87, 2024-08-22:
Ok, I have resinstalled everything now, still only the sensors are working, but the calculations of economy are not updating.

I believe this might be an issue with long term statistics then which don't get removed when I remove the underlying sensors.
Don't know...

Lets see if HA offers deleting long term statistics in bulk in the future, so I can remove everything related to Huawei and pees in bulk and start fresh

Thanks anyway!

JensenNick, 2024-08-27:
@pschneider87 I've made a short guide here [https://github.com/JensenNick/huawei_solar_pees/wiki/FAQ-%E2%80%90-Frequently-Asked-Questions#how-do-i-delete--clean--purge-previously-installed-sensors](url) on how to delete sensors properly including sensors in Statistics (history).

If you have issues with the two, you can try to provoke the startup by manually activating charge and discharge to/from grid - just momentarily 2 - 5 min.

JensenNick, 2024-10-05:
Reading through you posts again, I realize that you have set a fixed value as price input. This has to be a sensor that updates hourly. It is the price sensors that trigger the economy sensors marked _tt.

If you have fixed price to you can of course set them to trigger - but then everything could be made more simple.

closed 2024-10-05 by JensenNick

pschneider87, 2024-10-05:
I now use an input number with the static prices, and update it each hour with an automation. Seems to work out.