JensenNick/huawei_solar_pees #86: device_attr("sensor.inverter_input_power","model") does not exist

Mahai17, 2024-11-03:
Hi,
I am using huawei_solar (wlcrs) and huawei_solar_pees package.
Works great so far.
Now, I want to start using the "Efficiency Corrected Input", but that does not work.
In huawei_solar_pees_input.yaml line 190, you refer to device_attr("sensor.inverter_input_power","model") to determine the Inverter Model.
The sensor does exist but doesn't has the attribute "model". That leads to sensor.inverter_1_model being "unknown"

I have SUN2000-4.6KTL-L1

Regards
Martin

JensenNick, 2024-11-04:
Take a look here :-) https://github.com/JensenNick/huawei_solar_pees/discussions/85. Could you please provide the information about the output and I'll have a look at it.

Mahai17, 2024-11-04:
IndexError: list index out of range
[image]

I wonder, because there is no attribute "model" in entity "inverter_input_power"

JensenNick, 2024-11-04:
Support added in release v2.2.6.

JensenNick, 2024-11-04:
Support added with the release v2.2.6.